Silverquark/i3PlusPlus #5: Contrast

Bosbessenbasje, 2017-06-02:
Not really an "issue", but nevertheless... The contrast of the screen is a little awkward. Most of the time users will place the printer on a table top. When looking from above on the screen (when standing), the screen is (with the current color scheme) badly readable. Perhaps that was the reason of Wanhao to use the black/white screen settings.

I like the fact that colors are being used (why would you otherwise have a color TFT?), but perhaps the colors could be chosen a little bit different so that the contrast increases.

Silverquark, 2017-06-02:
Yeah the screen doesnt have very good viewing angles. Noticed it myself. Will try to make it more readable in the future. Im not a UI Designer and fear that if i change it too much it wont look good anymore. Thats why i decided to do a design similar to [googles material design](https://de.wikipedia.org/wiki/Material_Design) since its relatively easy to do and looks good.

Bosbessenbasje, 2017-06-02:
Hi!

Sure. No hurry. I appreciate what you are doing. I would love to help, but I just don’t have the time…

m.v.g.

Marcel Boom
Senior Designer
Neways Technologies B.V.
Voltaweg 10, 6101 XK Echt
P.O. Box 84, 6100 AB Echt, The Netherlands
T +31 40 267 9310
marcel.boom@newayselectronics.com <mailto:marcel.boom@newayselectronics.com > , www.newayselectronics.com<http://www.newayselectronics.com>
Member of the Neways Electronics International Group

crocky111, 2017-06-03:
Very hard to see the bed and temperature while the printer is running unless you are sitting right in front of it. Probably need characters with a higher contrast/thickness... I guess it about colour?

Silverquark, 2017-06-03:
Wanted to change the font anyway. Make it a bit bigger and more bold. But havent gotten around to figure out how to do it