Silverquark/i3PlusPlus #6: Some small things and suggestions

Tieya, 2017-06-02:
When PID tuning it can give a more stable temperatur if the fan is on while its running.
Maybe there can be added a button to turn the fan on at 100% or off.

also i see that the steps for the X and Y are set to 81 in the firmware that will result in parts that are to large. print something that is 100x100 and you will see what i mean.
i calculated the real step per mm to 80 using information such as belts-gear and steppers.
they set the wrong values in the factory.

thank you for using the time to make a custom firmware and sharing it (:

Silverquark, 2017-06-02:
Good idea! I wanted to look into how to make togglebuttons or checkboxes anyway. For example to set move menu step sizes to more accuratly be able to move the hotend. (also temperature adjustments). Will add it once i figured out how to do it. I'm currently printing a testplane with x,y on 80 and will compare the results. If they are better i will change it! I'm not a 3dprinting pro to be honest, only had the printer for a few months, so im always thankful for feedback of people who know more about the technical side 👍

jamesarm97, 2017-06-02:
Just found this firmware. I had started getting ready to overhaul the lcd firmware but now I guess I don't. Will see what I can do. I haven't looked around much yet but I know the nozzle preheat was a big issue I had. Needed a way to enter specific temperatures for different filaments. Original firmware would allow that, but only after a print was started which was stupid. Need to be able to heat the nozzle to any temperature before a print is started.

jamesarm97, 2017-06-02:
Shouldn't default steps per mm for Z be 400? I don't know where the 400.5 came from, originally Wanhao used 405 all over the place which was wrong. 200*16(microsteps)/8(mm per rev) = 400

Tieya, 2017-06-02:
jamesarm that is correct 400 is the calculated steps

Silverquark, 2017-06-02:
Not sure where the .5 comes from to be honest. This is a fork of nepes i3Extra he did the initial configuration and wrote much of the code before I joined the work. @jamesarm97 if you want to help out, check out our discord. Link is on the project Page. Will change it to 400.
You can use the 3 preheat profiles in the firmware. Later I will also add indipendent temp settings to the tempgraph screen

jamesarm97, 2017-06-02:
Ok, will try and check it out, too bad it isn't Slack. I hate using another chat app. Feel free to pop over to the i3 (and general printer) slack group some time also ;) I have seen a few plus users there. There was one other person really familiar with the lcd programming that was going to help port over the Repetier and I was talking with the developer of Cohersion3d about the mini board and using this LCD. https://3dprinterwiki.typeform.com/to/AgZRMx

Silverquark, 2017-06-03:
Wow Sounds interesting. Will definitely check it out. Would love to Talk to someone that has more experience with the lcd. Some things in it still remainin a mistery to me :D

Brandoskey, 2017-06-04:
I've never adjusted my steps for x,y and z and they're set to 81,81,400.5 respectively. That's probably why they were set that way. Are we sure the steps should be 80,80 and 400.5? This could explain some of my dimensions being a bit off.

Silverquark, 2017-06-04:
Beta 2 has These values fixed. 
Steps should be 80,80,400. Way more accurate results when printing big parts

closed 2017-06-04 by Silverquark