Silverquark/i3PlusPlus #7: Move menu during pause

chschmit, 2017-06-03:
Previously, when I paused a print, I was able to get to the move menu and jog the motors. This was helpful if I wanted to change filaments manually. Then I could go back to the printing menu and resume the print and the extruder would start where it left off.

Any chance this behavior can be restored? Or am I not seeing how to access the other menus while printing?

Silverquark, 2017-06-03:
No, you are right. It's not possible a the moment. I noticed that too and i will reenable leaving the print menu without stopping the print. Not sure how yet, since the free space for buttons in the print menu is pretty limited, but i will find a way

Silverquark, 2017-06-04:
Implemented in beta 2

closed 2017-06-04 by Silverquark