Silverquark/i3PlusPlus #8: PID acting strange.

Tieya, 2017-06-05:
Cant get the auto PID from the LCD to work anymore it just locks up.
works fine from pc.

can anyone else confirm this issue or is it just me? (: 

Version: B2

Silverquark, 2017-06-05:
No it's not just you. Beta 3 is coming today and will fix this and some other problems

Tieya, 2017-06-05:
thanks Silver.

damn just updated 30min ago :P damn you!

nice work!

Silverquark, 2017-06-05:
Sry,  im just human. Errors happen :D

dl6abm, 2017-06-05:
still don't work for me in beta3 ...

Silverquark, 2017-06-05:
Works fine here. What happens if you try it?
It takes a few minutes. It will display the temp graph while running and send you to a "done" screen after that

dl6abm, 2017-06-05:
works fine now, don't know why it didn'start the first time.

Thanks for your ( and the other contributors) work!
rgds Holger

closed 2017-06-13 by Silverquark