Silverquark/i3PlusPlus #10: Printing Speed setting

chschmit, 2017-06-10:
All of the printing settings (fan, bed temp, extruder temp) work except the printing speed.

I've tried running it from 10% to 200% and there is no perceived change in X/Y speed during the print job.
The odd thing was that the extrusion speed seemed to be effected by the setting. At lower speeds the nozzle under extrudes and at higher speed settings the extruder would over extrude. Its as if the X and Y (and maybe Z) speeds are unchanged, but the E speeds are.

jamesarm97, 2017-06-11:
Most firmware has Speed (like you are expecting) and Flow. Sounds like the Print speed is really Flow. They should both be options eventually (I would think).

chschmit, 2017-06-15:
Is the printing speed setting issue related to the new Marlin Firmware or something in the LCD firmware?

Silverquark, 2017-06-15:
You are right, it only affects the feedrate not the printing speed. But that's actually how it is in the original firmware as well... 
Will adjust the text from "printing speed" to "feedrate" for next version. Will also look into adding print speed as an extra option but cant promise anything

chschmit, 2017-06-15:
That's surprising, since I used the print speed feature with the native firmware and the XY speed would change and I didn't notice any under/over extrusion. It certainly appeared as if both speed and flow were varied proportionally.

Until, then I'll need to run the printer off usb since I frequently change speeds during the build and I found I can still do this with S3D. It's printing speed control varies both speed and flow. There is a flow control which can be used in parallel with the printing speeds but it's a percentage of the speed setting.

jamesarm97, 2017-06-15:
But Flow is automatically handled by the firmware if speed changes. The separate flow should be there to allow for under or over extrusion and the speed should be there to increase or slow down the printing speed. The lcd should treat them as separate items and the flow isn’t usually adjusted on the lcd just because speed changes (that is a firmware thing and is done behind the scene.)

Silverquark, 2017-06-15:
Never mind. found the issue. was setting flowrate instead of feedrate. fix will be included in next release

chschmit, 2017-07-25:
When will the next release be ready?

Silverquark, 2017-07-25:
Cant really say. I really dont have much time to work on it atm :/
Also my printer is currently broken...

closed 2017-09-04 by Silverquark