Silverquark/i3PlusPlus #11: Homing anomoly

chschmit, 2017-06-14:
After a printjob is completed the head goes to the near edge of the Yaxis and the midpoint of the Xaxis. If I run a new printjob afterwards (without turning off the printer), the printer uses this endpoint as the new home position and will be off by about 60mm in the X axis.

My starting Gcode has a G28 code that worked great with the original Wanhao (Marlin 1.0) firmware. If I use the motor controls and press the Home | All button it seems to override this position and homes it's self based on the limits switches OR if I power-off/power-on the printer, my Gcode files (with G28) work normally.

It seems there may be something preventing the G28 code from being accepted.

Silverquark, 2017-06-14:
Make sure you dont have a g92 in your end gcode

chschmit, 2017-06-14:
I double-checked...My ending gcode does not have a G92 command.

Since it worked fine with the previous firmware, I was curious if the new firmware handles homing differently.

Silverquark, 2017-06-14:
Hm. It's probably related to the marlin 1.1.x Firmware changes somehow. 
Had the same problem for a while but removing g92 from my end gcode fixed it for me. Will look into it

chschmit, 2017-06-15:
Is the printing speed setting issue related to the new Marlin Firmware or something in the LCD firmware?

vger74656, 2017-07-09:
I have the same issue mate... I have to tell the Octoprint to send X and Y to home, then print, otherwise it hits the right side and grinds...

jamesarm97, 2017-07-10:
Homing X after a print does not work. Tried stock firmware and I3Plus+ so it must be related to Marlin.

Silverquark, 2017-08-02:
Closing this issue to merge it with a duplicate, will link to here in the other issue.