Silverquark/i3PlusPlus #14: Massive under extrusion

andyfarthing, 2017-06-17:
Beta 3 - seems to have a problem with under-extrusion on prints, but seemingly at random. Tried about 6 or 7 prints, and only 2 succeeded.

I spent a day playing around with settings both in firmware and on Cura, but nothing seemed to have any effect. At random, a print would turn out okay even though I didn't make a change to any settings (just playing around with bed level).

I can push filament through manually fine, so pretty sure there isn't a blockage. Hitting the 'extrude' button in OctoPrint also works fine - it just seems to be during an actual print.

Reverted back to the original Wanhao firmware and problem went away instantly.

Silverquark, 2017-06-17:
Hm weird. Maybe a coincidence? Havent heard that from anyone else and never experienced it myself. The actual print Code also didnt change at all from beta 1 to beta 3. When does The print fail? On first layer? If so then it's probably just a leveling issue

andyfarthing, 2017-06-19:
It's possible - I'll run a few more prints on stock to see if it pops up again. If I can't reproduce it on stock then I'll flash back to this and see what happens.

They all failed on the first layer, so you could tell straight away. It looks like it would with a clogged nozzle, but I don't see how I've managed that if it is! I've ordered a new nozzle to try, just in case.

I did see a mention on the Facebook group that somebody else had an under-extrusion problem after flashing, which is why I thought it worth mentioning. They didn't comment with a follow-up though, so don't know how that progressed.

I'll run some more tests tonight and let you know how I get on.

Silverquark, 2017-06-19:
Its probably just the nozzle being too close to the print bed.
Where did you see that?

miikael, 2017-06-19:
I have somewhat similar problem.
I printed few test items and saw that there were some strange problems in prints. Sometimes underextruding and sometimes overextruding (left blobs of filament in random places).
So i tried to to e-step calibration. Sent over octoprint the command to extrude 100mm of filament, but it actually extruded only about 40 mm. So calculated new estep with the new result.
After applying the new estep number, i got almost same result with the command to extrude 100mm of filament. (little bit more filament was extruded, but not much)

andyfarthing, 2017-06-19:
Facebook comment here: [comment](https://www.facebook.com/groups/WanhaoDi3/permalink/846763215474911/?comment_id=847010832116816&reply_comment_id=847027042115195&comment_tracking=%7B%22tn%22%3A%22R4%22%7D)

As an update - after a few successful prints the same problem has now popped up in the stock firmware, so I'm not really sure what's going on. I've triple checked the bed level, replaced the nozzle, PTFE tube, tried different temps, different slicers etc. and no change.

Does anything carry over in the EEPROM when flashing between the stock firmware and this one? I'm just wondering if I've got some gremlins somewhere that are lingering... I've tried the factory reset but that hasn't made an improvement.

I'll try doing the e-step calibration as above in a bit, and see what result I get.

Silverquark, 2017-06-20:
Hm weird the only thing that may get transfered between flashes is preheat preset values. But they are stored outside of marlins normal save/load range. So that shouldn't interfere with anything. 
Can someone with this problem try sending a "M221 S100" and see if that changes anything?

andyfarthing, 2017-06-21:
I flashed back to this firmware (thought I may as well) - the first few prints turned out fine, but then the problem appeared again. The only change I made between prints was to run the PID autotune - could that have anything to do with it?

Ran the estep calibration and had the exact same result as miikael - only about 40mm was extruded. Changed the steps to 118 (which is what it calculates as), and a tiny bit more than 40mm was extruded.

Interestingly, if I run the "G1 E100 F50" command (from the wiki) to extrude, the motor runs backwards and ejects the filament. Any idea why this might be?

Running "M221 S100" doesn't have an effect, unfortunately.

Silverquark, 2017-06-21:
That's really weird. Autotune shouldn't have an Effect on it at all. All it does is run the M303 autotune command with the given parameters. 
M303 C10 S200 U1 would be the command for 200°C.
About the wrong Extrusion. If you are in absolute move mode and you give The command G1 E100, it will not extruded 100 but rather try to go to position 100, just like x and Y axis would do. Try doing a G92 E0 before. That will gell the extuder he is ar position 0. Or switch to relative Mode with G91

andyfarthing, 2017-06-21:
It might just be coincidence, but that's the only change I made. 

If I run "M221 S350", prints look fine (maybe a little over extruding, but certainly closer to what I expect). If you look at the image below, the top part (and bottom outer part) is run with "M221 S100", and the inner bottom with "M221 S350". The two blobs are where I paused it - ignore those!

[image]

andyfarthing, 2017-06-21:
Running the G92 E0 command before extruding sorted out the backwards motor, thanks. It still only extrudes 40mm though!

Silverquark, 2017-06-21:
Im currently working rebasing the firmware to a fresh marlin 1.1.3 build. No idea if this can fix the issue. But it will definitely clean the code up a lot and may fix some things I forgot to add from the original marlin source. Once I got it ready I will provide a download. Should be tomorrow or the day after. Let's hope that fixes it

gururise, 2017-07-09:
> Interestingly, if I run the "G1 E100 F50" command (from the wiki) to extrude, the motor runs backwards and ejects the filament. Any idea why this might be?

Just want to chime in here.  I have a Wanhao Duplicator i3 Plus on stock firmware.  Yesterday I was trying to calibrate esteps and after issuing the "G1 E100 F50" command the first time, it extruded about 90mm, I adjusted esteps and re-issued the command and noticed the motor ran backwards and ejected the filament.  (THIS WAS ON STOCK WANHAO FIRMWARE).  Everytime thereafter that I issued the command, it would eject the filament.  The only way to fix it was to power cycle everything and then reconnect.

So the issue is also in the stock wanhao firmware.

chschmit, 2017-07-19:
When will Beta 4 be available? 
I'm running Beta 3 and still having underextrusion issues like mentioned above despite resetting all of the EEPROM settings that were used prior to Beta 3.
I've even gone back to the original Wanhoa firmware and the EEPROM settings work just fine. I've since gone back to the Beta 3 firmware and LCD update the underextrusion is again happening.

Deneteus, 2017-08-19:
Any updates on this under extrusion issue? I noticed it was under extruding and found your Github while looking for a solution.

johnboiles, 2017-08-19:
FYI Cura has had an issue since ~2.6 where the filament diameter defaults to 2.85 even if the machine definitions specify 1.75 https://github.com/Ultimaker/Cura/issues/2270

After a Cura update I had issues that looked exactly like @andyfarthing's but it turned out to be a Cura problem. When testing between i3PlusPlus and Wanhao firmware, make sure to use the same gcode.

andyfarthing, 2017-08-25:
I (think) I managed to resolve it in the end, but I made the schoolboy error of changing two things at once, so I don't know which was the solution unfortunately. I haven't had the issue since however, so it may be worth trying:

1) Wiped the EEPROM (I used the clear_eeprom example sketch in the Arduino IDE for that).
2) Rebuilt the entire firmware up with Marlin 1.1.4 - I forked the Marlin repo and copied over the i3 Plus changes from this repo.

To be honest I'm not entirely sure it was ever down to this firmware - I flashed back to stock at one point and, although it solved the problem for a bit, it re-appeared after a while. In that case it's more likely that wiping the EEPROM sorted it. Either that or it was all just a complete coincidence and it was a problem that ended up sorting itself somehow.

I did notice that this firmware (at the time) seemed to default to 3mm filament in the Marlin settings, so I had to go in and change that using the firmware tool in OctoPrint. There may have been something related there that was causing the problem (the change I made in OctoPrint may not have saved properly perhaps?).

It looks as though this repo has been updated to 1.14 now, so it may be worth flashing that anyway if it's still affecting you.

@johnboiles - I did see the issue with Cura when I was having the problem (and it coincided with that release!), but I tried a few different slicers and the issue was still there. Definitely worth checking though.

weidercs, 2017-08-25:
That worked for me too, cleared the eeprom and I was able to get back to 'normal' estep settings.  Going to run some test prints, but things look back to normal.  Thanks!

Deneteus, 2017-09-03:
That issue with the firmware showing filament is 3.0 I saw as well. @andyfarthing. I had also tried to calibrate esteps and it was under extruding with a default of 93.5 and not 96 per page 25 of the manual. Marlin is 1.1.5 is out as of August. Does it fix this issue?

andyfarthing, 2017-09-04:
@Deneteus Well, it's not an issue with Marlin as such - Marlin defaults to 3mm anyway by design - it's just that obviously our printers run with 1.75, so it should have been changed to that in configuration.h as the Wanhao default firmware does.

It's something we can change ourselves through OctoPrint or similar, and would probably be over-ridden in the GCode anyway by whatever slicer you use (but not in the case of Cura 2.6). Updating to 1.1.5 wouldn't fix that, because as far as Marlin is concerned it's set as it's supposed to be and there's nothing to 'fix'.

Same with the estep configuration - I found mine to be out by a fair bit from what was configured in this firmware.

It's probably unlikely, but it's possible that somebody flashing this firmware might not notice the 3mm default set in the configuration, not set the diameter properly in the slicer, and then run into similar symptoms.

I thought it had been changed in the source in this repo, but I've just had a quick look and can only see 3mm mentioned (#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00 in configuration.h). 

On a side note - I've also seen another version of this firmware in a different repo (https://github.com/i3PlusPlus/i3PlusPlus), so I'm not sure which is supposed to be the actively maintained one anymore? It's all getting a bit confusing! :)

Silverquark, 2017-09-05:
The #define DEFAULT_NOMINAL_FILAMENT_DIA 3.00 setting should only have an effect if you have a filament width sensor as far as i know, but i will double check that to make sure.

The https://github.com/i3PlusPlus/i3PlusPlus repo was a test i did to make the account an organisation, ignore it for now, if anything changes i will make sure theres no confusion as to what is the right repo.

Silverquark, 2017-09-07:
off topic: @andyfarthing i couldnt find another way to contact you, so im doing it here. I noticed you have a fork of my firmware and i really like your design! I would like to talk with you about maybe making it the official one (with some changes) since ive been planing on changing the design to give it more contrast and make it simpler overall. If you are interested contact me (silverthequark[at]gmail.com)

jamesarm97, 2017-09-07:
I would think the filament diameter must be used if Marlin has any type of dynamic diameter control feedback since usually the printer firmware doesn’t care because the slicer is the one that usually sets the diameter.

uksomebody, 2017-09-30:
Haven't tried this FW yet.  I have a Maker Select Plus.   

FWIW I have experienced the massive under extrusion problem even on the stock firmware.  I never figured it out,  no extruder skipping or anything.  It sometimes would just extrude a few cm instead of 10.   I did have a ton of problems with the nozzle clogging easily and generally not extruding well.  Installed the Micro Swiss yesterday which flows massively better.  Fingers crossed it lasts.

Deneteus, 2017-09-30:
When I went back to the stock firmware it didn't under extrude. I am also using Microswiss. I haven't recompiled the firmware yet since I am working on my CR10 upgrades currently.

xKoldx, 2017-10-21:
I also had the same issue. Changing filament size in Octoprint from 3mm to 1.75 did not resolve the issue. I also know it probably didn't matter because my slicer (Cura) was setting the size for me anyway. 
strangely enough, after trying to do an E-steps calibration my math gave me almost 225 steps per mm, definitely far off from the 94mm stock. i too was extruding only 30-40mm (can't remember the exact amount), i set it to 225 that and still had under extrusion problems. i reverted back and eventually gave up. i ended up restarting the machine, unplugging everything and hitting the "reset to defaults setting". i probably did the "Autotune" as well, but after messing with it a few more times, i swapped filaments thinking that was the issue. i recalibrated my e-steps value and ended up with 114, so not too far off but definitely strange. runs fine with my current settings now. 

Using the Monoprice Maker Select Plus

uksomebody, 2017-10-23:
I gave up and installed the micro-swiss all metal hotend,  which is a
massive improvement.

Filament flow is so much better than the stock hotend ever was.  Zero clogs
or under-extrusion problems since installing it.

Silverquark, 2017-10-23:
**For anyone who has this issue, please download and flash the new changes in the repo and tell me if the problem still occurs**
(Green "clone or download" button then "Download zip" here: https://github.com/Silverquark/i3PlusPlus)
Installation is the same as any full release

xKoldx, 2017-10-23:
will do. let you know tomorrow how it goes. Thank You!

Silverquark, 2017-10-23:
forgot to mention, that a factory reset will be needed after flash

xKoldx, 2017-10-23:
i can do that with M502 and M500 right? then just off and back on? or do i need to hold the reset button?

Silverquark, 2017-10-23:
M502 and m500 will have the same effect as just pressing the button in the menu.

The old version had a bug in the factory reset button that caused only M502 to be executed. So it reset the data in the printer, but didn't save it in eeprom. This is fixed now

xKoldx, 2017-10-23:
Compiled using Arduino and uploaded. Used the factory reset button on the LCD Screen. Printed the calibration cube. Everything is working great!

johnboiles, 2018-01-22:
Not sure if it's related, but I've noticed while running e-step calibrations that sometimes extrusion seems to stop way short. Typically I'll end up with a result that's ~<10 % off my current setting but every now and then extrusion will stop midway leaving me with ~40% less extruded than expected. Recently I did 5 e-step calibrations in a row and 4 of them were reasonably near target and one of them way underextruded. I'm using the following to extrude 100mm of filament in a minute (after manually setting the temperature
```
M83
G1 E100 F100
```

cyberhiker1965, 2018-01-22:
I believe the I3++ project is on hold. You might want to consider a similar firmware, based on Marlin 1.1.8 called ADVi3++.https://github.com/andrivet/ADVi3pp-MarlinRegards, David

taleyaJ, 2019-02-07:
I just ran into a similar issue  - what temps are you running at? I was able to trip it even on the estep calibration when I stepped it down to my habitual 197, but at 200 it runs fine.  Ran a Gcode that was having issue with a temp override and it's gone through without issue.  Annoying, as I've previously been running on this exact filament at 197 without issues.