Silverquark/i3PlusPlus #18: printer on insane speed? (causing y distortion?)

kromar, 2017-07-23:
i installed this package today and got some really crazy printer behavior, i did print a calibration piece with 95% infill and usually the printer is really quiet, however this time it was terribly loud when doing the infill to the point where i was afraid that the printer gets damaged!
i did another print to give it another test after verifying that the hardware is stable and this time i got horrible layer distortions (also had the same insane noise/movement of the printer) so instead of the calibration piece having a vertical outline it was about a 30 degree angled piece  (haha never seen that before)
[image]

so im not sure whats going on, i assume there have been some modification in the settings? i will compare the EEProm values tomorrow and see if maybe there is a extra digit somewhere that should not be there, but then im at the end of debugging without going into the code. 
so maybe you have some idea what is causing this

Silverquark, 2017-07-23:
Thats really weird. The only thing that i changed print wise is x,y and z steps since the default firmwares values were slightly off causing prints to be a bit bigger then they should be. [See here](https://github.com/Silverquark/i3PlusPlus/issues/6)

Can you flash the wanhao 2.18 printer firmware again and make sure that the problem still exists? You shoudnt need to also flash the screen firmware to test that. But after testing you should make sure to have both either on mine or original since mixing them up will cause features to break. (sd menu and print should be compatible though)

kromar, 2017-07-23:
i flashed the **Originali3PlusFirmware_v3_fixed_no_preselect.zip** that was in your repository since wanhao page is down and i was not able to get a .hex of 2.18 from somewhere else. 
anyways with that v3 i do not have the same issues, the printer is quiet again and produces accurate prints.

Silverquark, 2017-07-23:
Shouldnt matter if its usb or sd normaly. Are you sure it wasnt just coincidence and just that one print failed?

kromar, 2017-07-23:
Im certain it not just failed, i did  a lot of printing and tweaking and  the behavior of the printer was totaly insane with this firmware.
I will give it another try tomorrow, but im a bit scared to damage the printer with this:/ 
Unfortunately i cant provide any more information what could cause this but im quite  sure that it must come from the flashed software.
If you want me to test something specific let me know.

kromar, 2017-07-24:
so i flashed the firmware again and the same print did not cause the same effect as yesterday.. hmm strange. 
i noticed that the acceleration value is read as 0 in mater control but shows as 800 default in simplify 3d. when changing the value it still reads as 0 in mater control but reads correct again in simplify3d.
maybe there is a issue with the acceleration that was causing this in a specific scenario?

[image]

i will flash the LCD a bit later and see if the behavior returns then.

Silverquark, 2017-07-25:
Thats strange. Acceleration shouldn't be changed at all and should be in the same place in eeprom. Only thing in eeprom that's different is the 3 extra preheat values which I save and load last, so they shouldn't interfere with anything

kromar, 2017-07-25:
I flashed the lcd yesterday and was not experiencing any of those insane movements that i had the first time which is good, it would have been nice to be able to reproduce the issue.
Maybe it was a glitch from the first flashing or so... I will do some more prints in the evening and see how it goes.

kromar, 2017-08-01:
I think this can be closed, i have not experienced the same issue again and have been using it for a week woth lots of different configurations.

kromar, 2017-08-02:
well i guess i closed this to early, of course i got it today again after closing the issue... but i can clearly identify where it happens and i also have a gcode file that produced this issue on my printer.
[steps_midi_failure.zip](https://github.com/Silverquark/i3PlusPlus/files/1195164/steps_midi_failure.zip)

it looks like it comes from the solid infill layer that is layed down and it shifts in the y axis. again it sounded horrible when it happend and i think this will damage the printer if it happens more often. so i tried to do a printer restart to see if it goes away, and other wise will reflash the stock firmware to see how that behaves on this exact print.
if this never happens on the stock firmware, i guess it must have to be something with the updated firmware. would i need to take this to the Marlin development or would this be something that has been added here in this project?

Silverquark, 2017-08-02:
I doubt any change i made could be causing it. It may be a marlin bug, if it is it could also be already fixed. Cant hurt to tell them though or maybe find someone with a printer thats running newest marlin and let him test it

kromar, 2017-08-02:
ok will see how it behaves with the original firmware since the printer still almost explodes when with speeds down to 30mm/s and i was able to do infills at 50mm/s before without the printer almost collapsing and sounding like someone is throwing a table around in the room next door, haha.