Silverquark/i3PlusPlus #20: USB Printing screen

jram0421, 2017-08-02:
I'd like to be able to have a screen when printing over USB, similar to how my V2.1 is, if that's possible. I run Octoprint with all my printers so it'd be nice.

Silverquark, 2017-08-02:
I thought about something like that. You can always go into the print menu when printing over usb, so you can see the temperature graphs.  What excactly does your 2.1 Show when printing over usb? Does it show file Name and remaining time?

johnboiles, 2017-10-28:
Maybe a bonus of some kind of feature like this would be that it _could_ prevent LCD functions while printing with OctoPrint.

I borked a print last week I think by bumping the lcd. I was cleaning up my workspace near the printer, heard an LCD 'beep' sound, then the print head started ramming into the side of the printer :(

Silverquark, 2017-10-28:
I am planing on adding a "lockscreen" function that prevents inputs after some time like on a phone. Don't know when I get it done though, don't have much time atm...