Silverquark/i3PlusPlus #22: Home X sometimes fails

pigrew, 2017-08-02:
I'm using my own build of the beta3 firmware. Sometimes, the on-screen HOME X command fails.

1. Move x to somewhere near the middle of its travel
2. Turn on/power-cycle controller
3. Use the load filament command, and wait for a little filament to be extruded
4. Exit that menu, and go to the move menu
5. Click the home X button

What happens is that the Z axis moves up about 1 cm, and then  the X axis moves right about 1 cm. The X axis isn't homed.

Can anyone else replicate this?

The USB monitor lists "G1 E1 F120", "G90", "G28 X0" as being enqueued.

ukmook, 2017-08-02:
I have just done your steps and got the same response, Z moves upwards a 1cm and X moves 1cm away from home.

Silverquark, 2017-08-02:
[Duplicate](https://github.com/Silverquark/i3PlusPlus/issues/11)
Closed other issue since this one is more specificly descriping the problem and has reproducable steps.

Probably related to newer marlin version. Will try to merge the fixes up to 1.1.4 of marlin when i find time to do it

TachyonStream, 2017-08-07:
I have this same issue with the stock version of the MakerSelect and Wanhao firmwares.  

As a workaround, If you disable the steppers before homing, it works as expected.  It must do something different if it already thinks it is homed.    I've added this to my GCode for starting prints and resuming after a pause (for filiment change), and it works well.  I do still see the issue when using the lcd for homing/leveling.

cloyd1815, 2017-08-08:
Probably a cheap way to fix it but if you add a M84 command at the beginning of the home X function you could solve this problem from multiple angles. Not sure if adding it there would be destructive to other functions.

mattcaron, 2017-08-09:
I can confirm that I also have this problem, both with stock Monoprice firmware and i3Plus+. Also, disabling steppers then rehoming seems to always work. So, it's definitely a widespread issue.

Deneteus, 2017-08-19:
I also had this issue and I had never seen it before until after I installed Beta 3. I was running Wanhao V09 before that.

johnboiles, 2017-11-26:
Same issue here on a version of master from a few weeks ago. Thanks for the detailed repro steps @pigrew