Silverquark/i3PlusPlus #23: Typo in temperature.cpp

Mechotronic, 2017-08-02:
There seems to be a truncated variable in this line. I get a not in scope error. It looks like it was accidentally trimmed from beta 2. Not sure about the variable names so I'll hold off on a pull right now. Should Be THERMAL_PROTETCION_PERIOD instead of _PERIOD

 thermal_runaway_protection(&thermal_runaway_bed_state_machine, &thermal_runaway_bed_timer, current_temperature_bed, target_temperature_bed, -1, 
        
_PERIOD, THERMAL_PROTECTION_BED_HYSTERESIS);

MartinEderer, 2017-08-02:
please check this commit
https://github.com/Silverquark/i3PlusPlus/commit/b3ba3a21a8fcf59a06180d28820fdcb463091339

closed 2017-08-02 by Silverquark

Mechotronic, 2017-08-03:
Sorry about that. I'll check the commits next time.