Silverquark/i3PlusPlus #26: nothing is working

blakeavandermeer, 2017-09-03:
ive just updated to beta3 on a maker select plus.  screen works, cannot start print, shows bed temp at constant 250 degrees.  after a couple of minutes nothing will work.  drives are disabled.  any ideas?

Silverquark, 2017-09-03:
Make sure all cables are inserted properly and try flashing the cocoon stock firmware from here:
https://github.com/Silverquark/i3PlusPlus/blob/master/OriginalFirmwares/OriginalCocconCreateFirmware_fixed.zip (as far as i know theres no stock firmware available for the maker select plus, but the cocoon should be pretty much the same)

blakeavandermeer, 2017-09-03:
well.  cocoon software seems to almost get me back to normal.  heated bed isn't working at all.  still cant print over 260 degrees for nylon.  has the same temp shutdown as the monoprice firmware.  
I did notice something last night before giving up, wanhoa stated in their instructions that the micro sd slot was only capable of an 8gb card.  I happen to be using a 16gb.  could this be part of the problem with getting any of the firmware to work?

Silverquark, 2017-09-03:
it normaly just failes to flash on some higher capacity cards, its worth a try i would say

blakeavandermeer, 2017-09-03:
ok.  heres an update.  re-installed plusplus firmware with 8gb card.  fixed a broken heat bed thermistor wire.  completed auto pid and now am printing with 70c bed temp and 260c hot end.  so far so good.  I think were good to go.  no issues at the moment.  ill let you know if anything comes up.  thanks for all of your hard work.  without your efforts id be unable to print nylon with this machine! :)

Deneteus, 2017-09-03:
The Wanhao Plus firmware starts at 2.15 and goes to 3.0. The maker select
just has different images and animations. I just received a firmware from
them that allows hotend temp of 280c because the stock firmware is set to
shutdown autotemp at 255c. I had been going back and forth for 2 weeks with
not being able to print abs. I had to use the downgrade firmware to erase
the lcd and then reload it because the one on their site was hanging.

jamesarm97, 2017-09-03:
Just make sure you are NOT using the stock hotend at those temperatures.

Silverquark, 2017-09-03:
Good to hear everything is working now again!
@Deneteus can you send me the firmware? Ive contacted monoprice and asked for the firmware but never got a response... I would like to add it to my original firmware Collection for people to be able to revert back to it when they have problems with - or dont like i3plus+

closed 2017-09-07 by Silverquark