Silverquark/i3PlusPlus #28: Code / Build Instructions?

GeoffMaciolek, 2017-09-20:
Hey! It looks like most of this repository is binary data; is there a codebase that's being used to build this?

And, are you able to provide build instructions?

Thanks!

Silverquark, 2017-09-20:
Hey, install instructions can be found here:
https://silverquark.github.io/i3PlusPlusSite/instructions.html

If you are refering to the files in the lcd firmware, theres a Tool to modify them called DGUS Sdk sadly the official Website seems to be down and i Wont be at home for another 3 days. I will provide a mirror link then. 
You can find more Information here:
https://github.com/andrivet/Wanhao-i3-Plus/blob/master/README.md
And I will add some Info and links to the Readme when im back home. Hope this helps for now

Deneteus, 2017-09-20:
Here is a link to their site. 

http://www.ampdisplay.com/download1.php?cat=HMI%20UART(DWIN)&sub_cat=DGUS

GeoffMaciolek, 2017-09-23:
The DGUS site is helpful, thanks!  I can now see the that "Marlin" folder contains the actual firmware.  I imagine that can be built the same as it is usually done?  (A link to a guide for that might be helpful).

How will this work going forward - do you think the Marlin changes can be distilled to a patch/set of patches relatively easily, to be usable with future releases of Marlin, or will we (the users) have to wait for you to update this repo?  (I imagine there's a fairly easy way for me to find these DIFFs, but I'm not actually sure)!

I'm just hoping to see this stay maintainable, rather than being unusable with newer Marlin releases, as is often the case with forks.  (Maybe that should be a new issue...)

closed 2017-09-23 by GeoffMaciolek

Silverquark, 2017-09-26:
You can build the Marlin firmware part with the Arduino IDE. The install instructions on the project web page contain detailed steps.

Im planning to continuos to update and support the firmware for some time. Although I dont have much time to actually work on it, so Updates Wont be as frequent.
The firmwares codebase is 1.1.4 atm so it's just slightly behind official Marlin. It also has some smaller fixes and updates in it (for example motor Step Menu). I will release an update soon, i would like to find a fix for the x homing anomaly before that though, which I havent been able to yet.

I also plan on wirting up some sort of documentation, that will explain what the firmware is made up of. Especially about the workings of the lcd screen and how the communication between the marlin firmware and the lcd works

Deneteus, 2017-09-26:
@Silverquark I think the X homing issue has been fixed in bugfix 1.1.x version. I am in the process of compiling a headless version for my CR10. I plan on enabling UBL on my i3 Plus as soon as I am done whitling the features down that I don't need.

mattcaron, 2017-09-27:
My solution for the homing issue was to put an M84 at the start of my GCode, because it doesn't happen if you kill the motors before homing.

Silverquark, 2017-09-27:
I tried pretty much the same in Code. Executing a m84 before every home, which didnt help at all for some reason. Maybe when adding some delay between the 2 commands

@Denetus
I3plus+ is now based on marlin 1.1.4 but I still get the problem when loading/unloading filament and homing afterwards