Silverquark/i3PlusPlus #29: Distill to "Patchset" for future Marlin releases

GeoffMaciolek, 2017-09-23:
Coming up with a patchset / patching instructions to turn a stock Marlin release into an "i3PlusPlus" release might be helpful, especially for future releases.

mattcaron, 2017-09-25:
Why would anyone do this? Just merge the changes in. When everything is in git, distilling the changes to a patch set which you then apply to another branch is just working against the SCM.

(Speaking as someone who maintains changes to something like 175 upstream FLOSS repos for the custom distros I build on my day job).

GeoffMaciolek, 2017-09-25:
I definitely defer to @mattcaron on this one! My SCM skillset is decidedly limited (beginner). I see no reason not to mark this as closed (invalid), if this functionality is straightforward with git, which it appears to be.

closed 2017-09-25 by GeoffMaciolek

GeoffMaciolek, 2017-09-26:
Is there a (relatively) straightforward way to see what this looks like in the Github interface? It appears this version of the Marlin firmware was forked from [nepeee/i3Extra](https://github.com/nepeee/i3Extra), so there's no apparent "direct" relationship to [MarlinFirmware/Marlin](https://github.com/MarlinFirmware/Marlin).  (This is likely something I could figure out myself, but if there's a static URL that would show the comparison, it might be nice to have in the docs. But, given this is likely a straightforward thing that any dev should know how to do, it'd likely only be of interest to quasi-devs like me).

Silverquark, 2017-09-26:
You can directly compare it to Marlin 1.1.4, the codebase recently got updated. The differences will be the changes that make up i3plus+