Silverquark/i3PlusPlus #30: screen brightness

Austinthemighty, 2017-09-27:
can you add a way to change the brightness of the screen at night it lights up my whole room.

Silverquark, 2017-09-27:
I was playing around with a brightness "slider" a while ago but couldn't get it to work. I should be able to make buttons for different brightness presets though. Will play around with it, may get it ready for next release, no promises though

Austinthemighty, 2017-09-28:
could you do buttons to change the brightness like "+" value "-" I think that the slider was throwing it off.  Also the brightness value may be from 0 to 255, and if thats the case multiple the value by 2.55 that way the brightness is based off of a 0-100%

cyberhiker1965, 2017-09-29:
You'd want to avoid the extremes, wouldn't be good if the screen is too bright or too dark to read 😁

pigrew, 2017-09-30:
This isn't a feature that I'm particularly interested in, but if it were added, I'd request that there be a "confirm" phase where if an "accept" button isn't pressed, it'll revert to its previous setting. It may also be a good idea to also provide a way to reset the brightness to some default, for example maybe pressing a particular part of the touchscreen during boot (or at least a M-code command).

Austinthemighty, 2017-10-06:
@cyberhiker1965 my printer is in my bedroom and at night with the printer on it makes falling asleep difficult to do as my whole room is lit up being able to turn the screen off without unplugging the printer would be great as I sometimes do 36-48 hour print jobs.  Also, there is currently a feature that when you don't press the screen for x seconds it will dim the screen and when you touch it, it will brighten back up the part that I want to be changed is when it dims on it on.

Silverquark, 2017-10-06:
Theres a workaround you can do at the moment. In the CONFIG.txt in the LCDFirmware/DWIN_SET directory you can adjust R6 and R7. R6 is the normal brightness of the backlight and R7 is the brightness when dimmed. They both accept 00-40 in hexadecimal (which is 0-64 in decimal). Where 40 is full brightness and 00 is no backlight at all.
Also you can adjust R8 which is the time before it dims the screen. this accepts 01-FF (so 1 to 255 in decimal), i believe this is in seconds. After you edited the file, you will need to reflash the lcd part of the firmware for the changes to take effect.

If you decide to turn the brightness off completely when the screen dims, be warned: the first touch will already trigger buttons, so you may accidently stop a print or something while "waking" up the printer.

I tried to find a way to stop the first touch from triggering buttons a while back, but couldnt find a reliable way sadly.

Austinthemighty, 2017-10-10:
ok thanks for the info i will give it a shot this weekend