Silverquark/i3PlusPlus #33: Beta 3 doesn't work with Maker Select Plus

kellogg76, 2017-10-21:
I flashed via Cura this morning and after it reported as a successful update I rebootied the printer where the MP boot logo appears and then freezes.

Reflashing the Wanhao firmware fixes the issue other than the language constantly changing to Chinese.

Silverquark, 2017-10-21:
It sounds like you didnt flash the lcd part of the firmware. Please follow this guide:
https://silverquark.github.io/i3PlusPlusSite/instructions.html

The language changing to chinese is an old problem of the wanhao firmware. They use the marlin acceleration value to store their language setting in it. Older slicers didnt change this value, but newer ones do, so the problem only started to appear recently.

closed 2017-10-21 by Silverquark

cyberhiker1965, 2017-10-21:
Did you flash the screen as well? Check the installation instructions - there are two updates: one for the control board (which you flashed via cura) and a second for the screen display (flashed via SD card, have to remove the screen to get to the internal card slot).

kellogg76, 2017-10-21:
First: Apologies I misread the FAQ, I thought it implied the firmware update enabled the LCD update to be done without disassembly, but now I see that it means future updates after the initial one can do that.

However, after copying the LCD files to the SD card and removing the LCD, there's still no update happening. The screen does not show any images, it lightens slightly so I know it's getting power but nothing happens other than that.

MicroSD was formatted on a Mac as FAT if that makes any difference.

johnboiles, 2017-10-21:
For me the Mac FAT formatting did not work either.

I think what I did was flash a Wanhao lcd firmware image onto the SD card, then delete all the files, and copy over the i3plusplus firmware files.

It would probably also work if you formatted as FAT from windows.

Silverquark, 2017-10-21:
You need to format it as FAT-32 and choose 4096 Bytes for sector size. (don't know how you can choose sector size on mac) 

Also if you use a >4gb microsd, it is also known to cause problems sometimes

dl6abm, 2017-10-21:
if i remember correctly i used a 4gb sd card. you need to format with a smaller cluster size  which i only could do on a windows platform (i used vmware fusion with w2k) 
rgds, holger

Silverquark, 2017-10-21:
@dl6adm
If that's true, it's really interesting. Will do some tests

kellogg76, 2017-10-21:
All good now, had to use a 4Gb card formatted on Win10 with 4096 bytes.

johnboiles, 2017-10-21:
I was able to make it work with a SanDisk 8gb SD card.

To @Silverquark's point i don't think you can choose sector size on macOS, so I grabbed the [8gb SD card binary image](https://www.dropbox.com/s/niy4e2aixs8uu4t/LCD217B_IMAGE.ZIP?dl=0) from [3dprinterwiki.info's factory files section](https://3dprinterwiki.info/wanhao-i3-plus-documentation-factory-files/) and flashed it onto my 8gb card with [Etcher](https://etcher.io/). Then you'll have a sd card with the correct formatting / sector size. Just delete all the files off the SD card and copy the i3PlusPlus files and you should be good to go!