Silverquark/i3PlusPlus #34: Is there a way to tune the e-steps?

rippledabs, 2017-10-25:
I love the way the new interface looks, but I can't find the motor tuning option. Is it somewhere?

Silverquark, 2017-10-25:
In the current beta its disabled cause it was bugged. 
If you press the green "clone or download" button and "dowload zip" however you get the latest changes, where its fixed and enabled again

johnboiles, 2017-10-28:
@Silverquark are you pretty confident in the master branch? Do you typically run master on your printer(s)?

Silverquark, 2017-10-28:
I'm always running my printer on the latest changes I made. I'm actually close to posting an update, but there are some things I want to include, which aren't in yet.
Would be nice if you decide to flash the master branch and report any issues you find.

johnboiles, 2017-10-28:
Cool, i'll load up master on an SD card and probably flash Monday when I should have some new bearings coming in. Thanks for all your work on this @Silverquark!

sniperxnl, 2017-11-11:
@silverquark, firmware almost ready?