Silverquark/i3PlusPlus #35: LCD Firmware not completing

agentnoise, 2017-10-26:
Monoprice Maker Select Plus. 

I have an 8 gig MicroSD card formatted to Fat32, allocation size 4096.  The LCD will load all the images, you can see the Thermal Runaway image flash by quickly, I believe that should be the last image.  However the LCD never shuts off, it just hangs at a white screen.  If I try to boot the printer after that it play the first second on of the Monoprice boot screen and does not proceed.   Wanhao firmwares load just fine. 

It seems like the config file is not being read and the STOP_DWIN_OS; is never called.

Silverquark, 2017-10-26:
Do you have a smaller SD card? Cards bigger then 4gb sometimes cause problems

mattcaron, 2017-10-26:
If you don't have a small enough card, making one with a 1GB partition worked for me.

agentnoise, 2017-10-26:
I just tried a 4 gig card and it had the exact same issue.  I will try to partition 1 gig off

johnboiles, 2017-10-26:
There [have been reports](https://3dprinterwiki.info/wanhao-i3-plus-documentation-factory-files/) of it helping to downgrade to wanhao's official 2.11 firmware before flashing new LCD firmware. Maybe worth a shot if the card partitioning doesn't help.

agentnoise, 2017-10-26:
The 1 gig partition did not work, exact same issue. Goes all through the images, then does not shutdown.   Unfortunately  Wanhao's 2.11 directory 404s. Trying to find another one.

Silverquark, 2017-10-26:
Hm this is weird, haven't heard of anyone with this problem before. If you find a 2.11 firmware can you send it to me? 
@johnboiles interesting, didn't know that, if we find a 2.11 firmware, I will add that trick to the faq

agentnoise, 2017-10-26:
I cant find one.  I have been digging and digging.  If anyone has a copy I would love to test this.

agentnoise, 2017-10-27:
Good news everyone! I asked Wanhao for the original firmware and they provided it to me wit the following message 

> WE suspended the link due to when the users download it file gets corrupted. If anyone needs to flash the firmware we can send the a SD card for 5$ plus shipping fee for 23$ and you can flash the firmware. 
I attached the firmware file to this email, you have a try first and see if there is any luck.

So take that for what you will.  I flashed my LCD with that and then flashed the Plus+ firmware.  It stayed on the bright white screen but worked when I rebooted the printer.  Some of the controls are  being odd though.  When the printer is first turned on, if you go to move and try to move any axis, it has to home it first.  Then it works fine and keeps working until the printer is rebooted.  This might all be because I changed some setting in Marlin for my Z probe though.  I will test more tomorrow

Original Firmware: [wanhaoPLUS新固件-V09.zip](https://github.com/Silverquark/i3PlusPlus/files/1420574/wanhaoPLUS.-V09.zip)

Silverquark, 2017-10-27:
Nice! Really interesing to see that this works. Will try to analyze the firmware to check if i find the reason it works.
The movement thing is intentional. The beta has a feature, that prevents the printer from going out of bounds and crashing into the frame. The downside is it has to home after a restart.
It is removed in the current dev branch, so it wont be in the next beta, since it can cause problems.
If you finish a print for example, then restart the printer and you try to move the print head away, it might crash into your print.

agentnoise, 2017-10-27:
good to know about the movement.  Huh, I didn't even check the version for the LCD firmware.  I asked for 2.11 and that's what they sent.  Good to know it works though