Silverquark/i3PlusPlus #36: Why M600 does not works?

sniperxnl, 2017-11-10:
The title say it , i use on my other printers a lot M600 , but i cant get it worked on the latest firmware

sniperxnl, 2017-11-14:
I found it in the firmware and i see it is off
When i activate it i get this error
#error "FILAMENT_CHANGE_FEATURE currently requires an LCD controller."

mattcaron, 2017-11-14:
That's probably why it's off then. The Plus doesn't have an LCD controller, at least not in the traditional sense. What it has is a "smart" display which sends commands to the printer, not unlike an Octoprint server.

sniperxnl, 2017-11-14:
This error is happening when i compile the config file in Arduino. I need this code for the work what i do with it

mattcaron, 2017-11-14:
Right, and, unless you go through and fix it so that:
1.) It compiles
2.) It works

It's not going to. My guess is that upstream established that dependency for a reason, so, even if you comment out the compilation error, it still won't work.

sniperxnl, 2017-11-14:
That is realy F* upped, i buyed this printer 6 days ago, so i can use M600 code. This is a pain in the ass then

weidercs, 2017-11-14:
This may help if you're using octoprint... https://github.com/MoonshineSG/OctoPrint-MultiColors

sniperxnl, 2017-11-14:
@weidercs  Nope M600 code between layers is much faster to use. Need to think about using Octo, anyway thnx for the tip

mattcaron, 2017-11-14:
So, let me get this straight - you're mad that third party firmware, that you pay nothing for, that the guy who maintains it is in no way compensated for, doesn't have a feature that you want?

If you want it to work, you have the code. Feel free.

If you want to pay silverquark to do it, feel free.

Otherwise, be happy that they've done the work they've done.

sniperxnl, 2017-11-15:
Sirieusly you think i mad on thr maker of this great upgrade? SORRY read it again. I am disappointed that i see it to late that it is not working on this printer. He did a great job ons this firmware. So don't make things bigger then they are

Silverquark, 2017-11-15:
mattcaron is right, the feature requires an lcd controller. It would be possible to change the dependencies by rewriting a bunch of code, so that it works with the lcd touch screen. Sadly i dont have time for that atm, since i started working full time again. 
[Andrivet](https://github.com/andrivet/ADVi3pp-Marlin) here on github is actively working on another custom firmware based on this one. Maybe he is willing to implement the necessary changes to make it work

mattcaron, 2017-11-15:
I read it again. Still seems you're mad at this project for not having a feature you want. Thank you for clarifying.

closed 2017-11-15 by sniperxnl