Silverquark/i3PlusPlus #40: Z axis home following a print

Crypticat, 2018-02-02:
Hi there,

I completed a print with this firmware loaded on my printer and for some reason the print head did not move to the dock position like it normally did. The print head just parked above the part. I needed to remove the part so I manually tried to move the Z Axis up. This seemed to trigger an auto home which caused the the nozzle to drive straight down into my part and jam up the steppers. Luckily my print head had cooled so there was no damage done to my part.

Would it be possible to disable this Z auto home after a print?

Thanks!

jamesarm97, 2018-02-02:
I would think all that you mentioned would be part of the ending gcode produced from your slicer. You should be able to add a G28 X0 Y0 to home X and Y but not Z at the end (unless the firmware is broken).

Crypticat, 2018-02-03:
My gcode ends with G90 then G1 X0 Y190 (My Y-axis is homes at the back of the printer so I manually drive it to the front). It worked with the older version of Marlin from Wanhao but for some reason it isn't working for this job I am busy with. I will try your suggestion for the X axis at least.